dslotter/HamPi #8: Create on-device installer for TeamViewer

dslotter, 2020-05-23:
I have created an Ansible playbook for installing TeamViewer, but cannot enable this in the release image because this is a commercial app and do not want to enable without their permission.

The legal workaround is to create an on-device installer for this so people can install TeamViewer for themselves.

Neon22, 2020-06-30:
Teamviewer is apparently very good at what it does. But if you get flagged for commercial use you might get stuck a bit. For a first hand report on this, scroll down half a page here:
- https://www.qrz.com/db/VA3HD

The suggested alternative is Chrome Remote Desktop. I do not know its status on the Raspberry Pi.

dslotter, 2020-06-30:
Interesting perspective. I think that VA3HD's was an edge case because he was using it on so many computers simultaneously. And also, TeamViewer is just an alternative to running VNC, which is what I actually use, as well as many others.