dslotter/HamPi #14: Add applications to  GUI interface that are installed

dranch, 2020-05-27:
Please consider adding the follow GUI programs to the main UI icon interface:
   - cw (xcwcp,xdemorse)
   -  sdr and analysis (gnuaisgui, inspectrum)
   - utilities (xplanet)

I'm sure other programs that are missing as well

dslotter, 2020-06-14:
@dranch I investigated these applications.

Candidates for me creating a .desktop file include: xcwcp, xdemorse, inspectrum and xplanet.

The one application which does not qualify is gnuaisgui because it requires gnuais to be launched separately and prior to it. I do not believe the Desktop File Specification accounts for this requirement.

dslotter, 2020-06-14:
@dranch, I performed some experiments and it appears that if you do not assign an icon in the .desktop file that the GUI (Gnome, LXDE, whatever) assigns a default icon which appears to be: application-x-executable.png

It seems to be bare-minimum acceptable but if you can find something better please let me know.

dslotter, 2020-06-14:
This has been addressed in git commit 456985e.

closed 2020-06-14 by dslotter