dslotter/HamPi #15: Remove duplicate icons from GUI interface

dranch, 2020-05-27:
Please consider removing a lot of the duplicate icons found in the GUI interface:

   - Programming (GnuRadio Companion, 

   - Education (fccexam, Gpredict, GTK Smith Chart Calculator, hamexam, Qsstv, SmartSim)

   - Internet (Gpredict, Direwolf, Fl-suite, Gpsk31, Gqrx, Linpsk, Qtel, Quisk, Xastir, 

   - Sound and Video (js8call, jtdx, Message Aggregator, psk31lx, wsjtx, 

   - Other (twlog, 

   - Accessories (Chirp, Glfer, Quisk, TrustedQSL, Tucnak2, xlog, 

   - HamRadio (Direwolf, fccexam, Fl-suite, Chirp, Glfer, GnuRadio Companion, Gpredict, Gpsk31, Gqrx, GTK Smith Chart Calculator, hamexam, js8call, jtdx, Linpsk, Message Aggregator, psk31lx, Qsstv, Qtel, Quisk, TrustedQSL, Tucnak2, twlog, wsjtx, Xastir, xdx, xlog,

dslotter, 2020-05-27:
Let's talk about this... The reason these duplicates occur is because the original authors have created *.desktop files with multiple menu categories, rather than just one. But on the other hand, this is a unique collection of software and the overall appearance is this project's responsibility.

dranch, 2020-05-27:
If I take it from an enduser perspective, I don't care how the software was written.  I would just expect things to be cleanly layed out and not have duplicate icons.  To start, I would recommend to remove this duplication yourself and in the future, I would recommend to create tickets with the upstream projects to remove this duplication

dslotter, 2020-06-16:
This was fixed with the changes for Issue: #13 so closing as duplicate

closed 2020-06-16 by dslotter