dslotter/HamPi #17: Consider removing unneeded or duplicate software

dranch, 2020-05-27:
I would recommend to REMOVE Mathmatica and Wolfram as that could free up a bunch of space.  

There is also a lot of other software in there that I question if it's needed like all the IDEs, Pascal, Ruby, GCC vs. LLVM, etc.

dslotter, 2020-05-27:
@dranch I can't find Mathmatica and/or Wolfram on the image. Can you please give me either their package names ("dpkg -l") or filesystem location ("/usr/bin/foo")?

dranch, 2020-05-27:
In the GUI, see:

    - Programming (Mathmatica, Wolfram)

These seem to be coming from:
-rw-r--r-- 1 root root 250 Jun 24  2019 /usr/share/applications/wolfram-language.desktop
-rw-r--r-- 1 root root 447 Jun 24  2019 /usr/share/applications/wolfram-mathematica.desktop

dslotter, 2020-06-03:
@dranch I can't find those files on the HamPi image. This is all I can find:

`pi@raspberrypi:~ $ sudo find / -name "*wolfram*"`
`/usr/share/rp-prefapps/wolfram-mathematica.png`

dranch, 2020-06-04:
Can you not see the two files I listed above?  I pulled those right out of my running Pi image after I had done a full apt update.

johnhays, 2020-06-07:
I tend to start with a base image of [Raspberry Pi OS (32-bit) with desktop](https://downloads.raspberrypi.org/raspios_armhf_latest) which does not load up with a bunch of applications that are not needed for base functionality.

dslotter, 2020-06-14:
@dranch Ok, I see what you're talking about. It depends on the base image I'm using whether Mathematica is installed. I will make sure this is removed from the final release.

dslotter, 2020-06-14:
This has been addressed in git commit 7b2ec23.

closed 2020-06-14 by dslotter