dslotter/HamPi #21: Add 64bit ARM support

dslotter, 2020-06-04:
Raspberry Pi Foundation released a 64 bit Raspbian image. See if it can be supported.

dslotter, 2020-06-04:
64bit ARM support isn't totally ready for prime-time. Necessary libraries are not available for components like WSJT-X and JS8Call and SDRPlay RSP.

Added as much support as I could to git commit 5085a63.

closed 2020-06-04 by dslotter

ra1nb0w, 2022-02-03:
Now it is [stable](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/) therefore HamPi can move to 64bit and get some speed enhancement. thank you to maintaining the project.

ra1nb0w, 2022-03-16:
I tried last night and for my personal usage (direwolf, fldigi, hamlib, gpredict) seems to working fine (on raspberry 4).

thank you very much!

bluntforcefive, 2022-03-16:
Can I just say how much I appreciate seeing the conversation come across my
inbox? It inspires me to continue learning and to share even my brief
journey, it if could inspire one person in the way this project and the
people behind it have inspired me, I will die satisfied..

dslotter, 2022-05-08:
I've laid most of the groundwork for a 64-bit version. I need to get VNC working and I should be able to ship it.

patrickdickey52761, 2022-05-13:
> I've laid most of the groundwork for a 64-bit version. I need to get VNC working and I should be able to ship it.

Hi, Dave,

If you're using the latest version of the Raspberry PI Os (using Bullseye), you might need to make sure that Wayland isn't running. There is an article on their site https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/ which talks about how they're using the Mutter Window Management instead of Openbox, and VNC doesn't work with it. But, supposedly, if you enable VNC in raspi-config, it switches back to the OpenBox Windows Manager. It might be something to check and file an issue with them, if necessary.

Have a great day. :)
Patrick.