dslotter/HamPi #22: Add AMBEserver service and configuration files

johnhays, 2020-06-07:
Create a user opendv
**sudo useradd -s /bin/false opendv
sudo usermod -G dialout opendv**

Please find the file ambesupport.tgz in **https://nw-digital-radio.groups.io/g/ambe/files**
It contains:
A service file which is looking for AMBEserver in /usr/sbin -- move AMBEserver to /usr/sbin or modify the ambeserver@.service file. 
Three configuration files which should be placed in /etc/opendv

You can 
**mv ambesupport.tgz /tmp && cd / && sudo tar -xzvf /tmp/ambesupport.tgz**

Use the normal systemctl commands to use the service, e.g.

**sudo systemctl status ambeserver@ttyUSB0**

More info at the [wiki] (https://nw-digital-radio.groups.io/g/ambe/wiki/1520#Quick-Build-on-Linux) post.

Attached is a zip file containing the .tgz
[ambesupport.zip](https://github.com/dslotter/HamPi/files/4743183/ambesupport.zip)

dslotter, 2020-06-07:
@johnhays How do we know that the AMBE device will get ttyUSB0 and not a radio (like an ICOM 7300) that is plugged in via USB?

johnhays, 2020-06-07:
We don't actually, but we can mitigate it using this information
https://rolfblijleven.blogspot.com/2015/02/howto-persistent-device-names-on.html

Here is an example for the ThumbDV
root@hampi:/etc/udev/rules.d# cat 99-usb-serial.rules
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6015",
SYMLINK+="thumbDV"

root@hampi:/etc/udev/rules.d# ls -l /dev/thumbDV
lrwxrwxrwx 1 root root 7 Jun  7 16:47 /dev/thumbDV -> ttyUSB0

dslotter, 2020-06-16:
@johnhays I'm just now getting to this issue, but technically with the menu changes done, we're in a code freeze for 1.0 release. I'm afraid this is going to have to wait for 1.1. But the good part is that ambeserver is compiled and present on the image, so all that has to be done is someone to run it -- or follow directions similar to these.

johnhays, 2020-07-09:
An improved version of the solution.
Create file  **/etc/udev/rules.d/99-thumbdv.rules**
Contents should be
**SUBSYSTEM=="tty", ATTRS{product}=="ThumbDV", SYMLINK+="ThumbDV"**

Create file **/etc/systemd/system/ambeserver@.service**
Contents should be
**[Unit]
Description=DV3000 Daemon
After=network.target serial-getty@%i.service dev-%i.device
Conflicts=serial-getty@%i.service
OnFailure=serial-getty@%i.service

[Service]
User=opendv
EnvironmentFile=/etc/opendv/%p-%i.conf
PermissionsStartOnly=true
ExecStartPre=/bin/chgrp dialout /dev/%i
ExecStartPre=/bin/chmod 0660 /dev/%i
ExecStart=/usr/sbin/AMBEserver -p ${UDP_PORT} -i /dev/%i $AMBESERVER_FLAGS
Restart=on-abort

[Install]
WantedBy=multi-user.target**
Create file **/etc/opendv/ambeserver-ThumbDV.conf**
Contents:
**#
# Configuration file for the AMBEserver program started by systemd.
# Adding flags to AMBESERVER_FLAGS will pass this to the daemon.
# For example, to change the speed to 460800, use the following line:
# AMBESERVER_FLAGS="-s 460800"
#

UDP_PORT=2460
AMBESERVER_FLAGS="-s 460800"**

---
I will make a *.tgz file that you can use as part of your build.