dslotter/HamPi #33: Wiki start

Neon22, 2020-06-23:
Happy to start in on the wiki.
Say one entry per included program by desktop folder maybe.
with a short description, version and program original online location and website if available.
we can add notes about the specific distro aspect  and what else it can connect to as we work that out more...

How does that sound ?
You just need to make the initial page and add people/permission (I think) as they request access to add to the wiki.

Cheers....

dslotter, 2020-06-23:
First page created at https://github.com/dslotter/HamPi/wiki

I don't see any settings for permissions. Please see if you can edit it now. Reopen this issue if you can't.

Neon22, 2020-06-23:
I can't edit yet. I think its to do with puublic/private repos.
some help here:
- https://help.github.com/en/github/building-a-strong-community/about-wikis
- https://help.github.com/en/github/administering-a-repository/setting-repository-visibility

ok. so here'e a possible approach. (setup a sep project which is public so anyone can contribute then push updates to the wiki on this project.) more details here:
- https://labs.inn.org/2014/05/19/applying-git-to-github-wikis/

SO:
 - could make project public - but anyone can change anything so probably not.
 - make sep wiki project and push to this one.
 - add people as team members or something but works only with small trusted group of people.

Neon22, 2020-06-26:
@dslotter   which of those last three is the one you think we should go with ?

Neon22, 2020-06-27:
I've tried cloning but can't seem to get the remote right for a wiki (which is diff to regular code) so I can push. 
The solution seems to be to add me as a collaborator:
- https://help.github.jp/enterprise/2.11/user/articles/inviting-collaborators-to-a-personal-repository/

Then I can try to push wiki stuff back to you. Note that I have cloned only the wiki and will not be pushing anything to files.
@dslotter  Do you want to try that ?

dslotter, 2020-06-27:
I have sent you an invite to collaborate. Please let me know if that works.

Neon22, 2020-06-27:
Have a look on the Wiki at the logging applications in sidebar.
let me know if hi sis the direction you want to go in.
I will also try editing the Home.md tomorrow.

dslotter, 2020-06-27:
I think that's an excellent start!

Neon22, 2020-06-27:
and we're off. more tomorrow. cheers.

Neon22, 2020-06-28:
In WinLink section we have several things by john.wiseman:
- http://www.cantab.net/users/john.wiseman/Downloads/Beta/
- ardopc
- piARDOP_GUI
- ARDOPOFDM.exe

There is pi gui program and a command line program and several others.
Which ones are installed in this distro ?

Neon22, 2020-06-28:
Is the antenna program xnec2c included ?
- https://www.qsl.net/5b4az/pkg/nec2/xnec2c/doc/xnec2c.html

Neon22, 2020-06-28:
I found the yagiuda link in Antennas is now linked to an aggressively  malicious site.
I contacted the owner and he will change the link on his website (he lost control of the .org domain) but this is the link we should now use for the Yagi-uda info:
- https://www.g8wrb.co.uk/yagi/

wiki updated.

Neon22, 2020-06-28:
Only a handful  left to do. I hope t finish tomorrow.

Notably there are so many forks of predict for satellite tracking. Python versions, gnome versions, etc etc... very active.

Neon22, 2020-06-29:
Its worth noting that the Wsprry fork was lastupdated in 2015.
- https://github.com/8cH9azbsFifZ/WsprryPi

more recent Wsprry fork would  be:
- https://github.com/threeme3/WsprryPi
which has raspberry pi4 support

Neon22, 2020-06-29:
worth noting that the dump1090 fork was last updated in 2014.
-  https://github.com/MalcolmRobb/dump1090

There was a later release 3.8.1 which is a common fork much further on than the Malcolm Robb one.
E.g.
- https://github.com/arn4/dump1090
  -  customized web interface. installed both dump1090-mutability and dump1090-fa.
  - which looks like this: https://uz.sns.it/~arna/progetti/flight-tracker/

- https://github.com/MartinRusk/dump1090

- https://github.com/flightaware/dump1090
  - fork of dump1090-mutability customized for use within FlightAware's PiAware software.

also do we have this:
> "FlightAware have released PiAware, their dump1090 client for FlightAware - see: https://flightaware.com/adsb/piaware/
> It’s a simple package that connects to an existing dump1090 instance and sends the data to FlightAware over an encrypted channel, authenticated with an existing FlightAware account.  It works well, and doesn't provide a significant extra load on the Raspberry Pi."

Neon22, 2020-06-29:
Is the acfx the 2019 version from here:
- https://www.freshports.org/comms/acfax
There are many older versions. This may be the most current ??

Neon22, 2020-06-29:
@dslotter  OK. All uploaded. Let me know if I missed anything, or if there is some way to determine all of the versions.

Trying to think of a way to determine if the wiki help gets out of date in a subsequent version when you upload new packages or new versions of existing packages.

Cheers...

Neon22, 2020-06-29:
Also wondering if you'd consider adding Antennavis and Yagilogper to the distro:

- https://savannah.nongnu.org/projects/antennavis/
- https://sourceforge.net/projects/yagi-logper/

Found while digging for the wiki help info.

Neon22, 2020-06-30:
@dslotter 
By trawling through the tasks folder I think I found some more packages. SDRsharp, airspy, BOINC.
At least two of which I list above as perhaps being needed. 

I am not convinced I found and documented all the binaries though. Do you know a way to find them ALL?
Is there a preferred kind of ordering you'd like ?

The recommendations above for:
- yagi-logper: https://sourceforge.net/projects/yagi-logper/
- piaware for ADSB: https://flightaware.com/adsb/piaware/
- dump1090 appears significantly out of date (but not certain)
- Wsprry appears out of date (but not sure if newer is better).

still stand. Cheers...

KE0ZUW, 2020-07-22:
> @Neon22 I have sent you an invite to collaborate. Please let me know if that works.

Dave,   Can you give me an invite as well.  Mark and I have been discussing this on a HamPi thread.  There is enough to do for two of us.  KEoZUW  John

dslotter, 2020-07-22:
@KE0ZUW Ok, I have sent you an invite to collaborate.

KE0ZUW, 2020-07-22:
I received it. Thank You.  I will be on later today!

John

Neon22, 2021-03-25:
You can probably close this. now. Do you want  the wiki updated for 1.3 ?
I never did end up getting any emails for the sister project you contacted me about. Am I on a list ?

dslotter, 2021-07-30:
@Neon22 Thanks for all you do. It helps a lot. Could you please update the wiki for 1.3 and 1.4a? You can get the release (change) information at https://github.com/dslotter/HamPi/releases

Neon22, 2022-05-30:
Updated for V3