dslotter/HamPi #38: Miscellaneous issues

f6bvp, 2020-07-01:
While Hampi 2.0 download links are all broken ...
here is a list of issues in distro 1.0

- partition size is not expanded at first install boot. raspi-config did it.
- GPU memory is so small at 64 
- dtparam audio on is not mentionned in file "Activating your DRAW"
Despite following instructions in this file DRAW is not recognized while it work in another NorthWest digital distro.

There are certainly good reasons why kernel is not up-to-date.

Thanks for providing such an interesting bunch of applications for hamradio.

73 de Bernard F6BVP / AI7BG

f6bvp, 2020-07-02:
i made a confusion between Hampi 1.0 and Hamradio Distribution v2.0.
I realize that the last one is actually a former distro.
This should be more explicitely explained on Groups.io RaspberryPi-4-HamRadio
Bernard

f6bvp, 2020-07-02:
What should I check in order to get DRAW activated ?

Neon22, 2020-07-03:
Are you still talking about Hamradio Distribution v2.0 ?

If not do you mean the tasks in the file:
- include_DRAWS_support.yml
referring to "This is for the D.R.A.W.S. / UDRC from NW Digital Radio"

or something in the pdf 
- Activating your DRAWS™ HAT.pdf
 in HamPi/files/home/pi/Desktop/