dslotter/HamPi #41: Debian HAM Radio Blend

PhilmacFLy, 2020-07-17:
First of all I haven't looked in all depths into the way the images are build, but from what I have seen a lot is installed manually.

To, maybe, make this a bit easier, I feel a look into the debian HAM radio blend <https://www.debian.org/blends/hamradio/> could be beneficial. It provides several ham based meta packages that automatically install a ton of HAM related programms from the debian repos.

dslotter, 2020-07-17:
First off, nothing is installed manually. Ansible playbooks run which install the software. And in the cases applications are installed from source, they are usually more current versions of the applications that are in the Debian repository.

closed 2020-07-17 by dslotter

PhilmacFLy, 2020-07-18:
I saw that ansible is beeing used, but building the packages from source is imho the most manual way of installing packages that exists, even if its automated with ansible.

Of course you make a strong point with the software being used directly from source is way newer but there is also some major downsides that come with that. 
One of them is that one gets a system that is hard to maintain update wise. On of the main beauties of linux towards windows is that I have a package manager in the system know which software is installed and which software needs to be update. If I now install software by hand I'm getting the same mess as with every windows pc.
Another downside is that with manually installing an compiling software from source, is conflicts with other packages. Debian package maintainers take great care that packages in a stable Debian distribution work well together.  If you take packages from other sources then you may find that they somehow do not work with the “old” version of other packages that you installed from the Debian sources. This can quickly lead to an update frenzy where you need to update several unrelated packages that in turn break other packages.
Another disadvantage, if though for this scenario a small one, is that there are no security updates for the software you compile yourself. In Debian’s security updates you will get patches for the existing software version. If you installed version 1.3.8 then you will stay on version 1.3.8 – just with security fixes applied. I'll give you that this is not a big point on a raspberry pi that is running in your local network or in no network at all, but it still nice to know that all software is nice and secure.
The last point I'd like to make is that with the stable branch you get exactly that. Long lasting stable software. Especially in a portable/mobile setup like the HAMPi project is aimed at, I personally rather have a stable non crashing version of a software instead of the one with all the bells and whistles that maybe doesnt function correctly.

I see where you are coming from and I just wanted  to give you my 2 cents on that matter. No hard feelings.