dslotter/HamPi #51: auto build on distro base

Neon22, 2020-09-05:
Dave,
Have you seen this ?
- https://forum.khadas.com/t/making-fenix-images-in-github/9296

Is it applicable ?
Can you use the idea to build HampI thsi way ?

I ask because I'd like to try HamPi on a Fenix base. My Pi4 has 8GB (assuming it ever arrives :))

dslotter, 2020-09-06:
@Neon22 What is Fenix? Can you send me a URL for more information, please?

Neon22, 2020-09-07:
Fenix (pronounced Phoenix) raspberryPi4 distro with 4 desktops with 3D switcher. 350MB
So you can (say) open sdr on one desktop, antenna design on a second and logging on a third...
runs well without overclocking.

Quick demo on youtube (8min)
- https://www.youtube.com/watch?v=bV9b5lZ8efM
or 10 mins but quicker to the demo
- https://www.youtube.com/watch?v=OCI5zXeDicg
or:
- https://youtu.be/23QGtHuH0lA?t=47

core website (translate from spanish)
- https://fenixlinux.com/

Neon22, 2020-09-16:
Dave - What did you think of Fenix ?

Neon22, 2021-06-13:
I think we can close this :)

closed 2021-06-13 by Neon22