dslotter/HamPi #52: please update readme.md to point to Wiki

Neon22, 2020-09-20:
So users can find it more easily. 
Apparently most people don't check to see if any given github  has a populated Wiki.

Neon22, 2020-10-05:
Hi Dave,
Docs updated to latest packages I could find in v1.1

dslotter, 2020-10-06:
@Neon22 , thanks for updating the Wiki. I've also added a link to the wiki from the main page -- Git Commit # 30d5000

Please note that you can find my changelog here: https://github.com/dslotter/HamPi/releases . Although right now it's showing 1.0 as the current release, mistakenly, rather than 1.1. Trying to figure out how to correct that...

Neon22, 2020-10-06:
trying to update wiki - somethings gone squiffy somewhere. wil ltry again later

closed 2020-10-06 by Neon22