dslotter/HamPi #56: Anti-Malware was triggered

tactiham, 2020-10-28:
Bitdefender detected a trojan upon download. 
Says: "found in: https://clients.futuremark.com...ationRibbon/navigationRibbonJs"

dslotter, 2020-10-28:
What link were you attempting to download from? I can't do anything without knowing that.

tactiham, 2020-10-28:
The GitHub.com download link on your HamPi repository page. The .zip file option is what triggered it. Also, downloading the .zip file on the raspberry pi prevents any function of the gui. Once downloaded to the Pi, the various taskbar icons would open for a brief second then close. As far as I could tell after downloading the .zip file the Raspberry Pi OS was bricked. It was repeatable after a fresh OS install on the Pi.

tactiham, 2020-10-28:
Screen shot from the Bitdefender central app, with personal identifying information removed 

[image]

dslotter, 2020-10-28:
Same question as before: What link were you attempting to download from?

Answer only with a URL please. :-)

HamPi is downloadable as a ".xz" file only. I include a hash file so you can verify authenticity. Here is the current sha256sum hash:

b6a799906f3dd88320df19ad2b82e57f93053e600a108255cb23a4bc1cf095bd  HamPi v1.1 by W3DJS/HamPi_v1.1.img.xz

I do not have a "zip file option" for downloading. So that URL becomes even more important to have. Because if there are fakes of HamPi running around, I want to know about it.

And you are misusing the term "bricked". "Bricked" means that the device has had its ROM corrupted and needs to be reset with a JTAG type programmer. Just because you can't boot does not mean you have a "bricked" device. HamPi has been downloaded over 23,000 times. If it were "bricking" anything, I would have heard about it well before now.

dslotter, 2020-10-28:
I found these zip files online for "hampi". This is not my software: https://code.google.com/archive/p/hampi/downloads

Also, can I have your callsign, please?

tactiham, 2020-10-28:
https://github.com/dslotter/HamPi/archive/master.zip

tactiham, 2020-10-28:
Yes. KM4HPK

Martin.

It seems that it isn't happening now after I re-downloaded it.

dslotter, 2020-10-28:
I finally see what you're trying to do. You are attempting to download the SOURCE code that BUILDS HamPi. It is not the HamPi image itself that runs on a Raspberry Pi. And it certainly will not brick your device.

To download HamPi image, go to this link: https://sourceforge.net/projects/hampi/

I strongly recommend you join this group for further assistance: https://groups.io/g/Ham-Pi

closed 2020-10-28 by dslotter