dslotter/HamPi #60: Requested additions to Wiki

dslotter, 2020-11-11:
@Neon22 , I've added three software packages to HamPi for version 1.2. Can you please add them to the wiki?

1. acarsdec - Issue #57 
2. rpitx - Issue #58 
3. leansdr - Issue #59 

Also, LimeSuite is included in the image, but is not documented. Its GitHub project is here: https://github.com/myriadrf/LimeSuite  -- Can you please document that as well? It lives on the image under /opt/install/LimeSuite

Neon22, 2020-11-12:
ok. working on it

Neon22, 2020-11-28:
All updated and added on Wiki

dslotter, 2020-12-06:
@Neon22 I was wondering ... what are your thoughts about going forward including the Raspberry Pi OS release notes with the revision notes for HamPI?

If you decide it's a good idea to include them, then here are the latest release notes from https://downloads.raspberrypi.org/raspios_full_armhf/release_notes.txt

2020-12-02:
  * PulseAudio now included and running by default
  * Bluealsa Bluetooth interface removed - Bluetooth audio is now handled by PulseAudio
  * LXPanel volume control plugin replaced with PulseAudio version
  * Version 84.0.4147.105 of Chromium web browser included
  * Version 3.3.0 of Thonny included
  * Version 32.0.0.453 of Flash player included - note that this will be the final release of Flash, as it is end-of-lifed at the end of 2020
  * CUPS printer system included, along with system-config-printer CUPS GUI and HP printer drivers
  * raspi-config menu structure rearranged to match Raspberry Pi Configuration tabs
  * Control for GPIO-connected fans added to raspi-config and Raspberry Pi Configuration
  * Control for power / activity LED on Pi 400 and Pi Zero added to raspi-config and Raspberry Pi Configuration
  * Improved screen reader voice prompts in several applications
  * Added ctrl-alt-space shortcut to install Orca screen reader at any point
  * Low voltage warnings added to battery monitor plugin
  * Magnifier plugin zoom can now be changed with scroll wheel when pointer is over icon
  * Change to notification popups - now will only close when clicked on directly, not by clicking anywhere
  * Bookshelf now made compatible with translated versions of books and magazines, and will offer translated versions where available, based on system language setting
  * Bug fix - crash in CPU temperature plugin when throttling detection fails
  * Bug fix - if Orca is running, shutdown commands and shutdown dialog will force kill it to prevent it locking up the reboot or shutdown process
  * Various additional language translations added
  * Various minor bug fixes and UI tweaks
  * Raspberry Pi firmware b324aea801f669b6ab18441f970e74a5a7346684
  * Linux kernel 5.4.79

Neon22, 2020-12-06:
If you think its a good idea - then I'll add them in.

I guess the downside is keeping them up to date  when you make a new release.
- Specifically ALL the your changes between the OS versions from prev release and your new one.

The other approach says your help files would point to a specific Raspberry Pi OS release and users can take it from there.

What would YOU prefer ?

dslotter, 2020-12-06:
Honestly, @Neon22 , I'll leave that decision to whatever you think is best. I trust you to make a good decision. :-)

dslotter, 2020-12-06:
@Neon22 In case you don't see the other issue, would you please document the new scripts added in Issue: #66  ?

Thank you!

dslotter, 2020-12-06:
@Neon22 Universal Ham Radio Remote is the last new application to add to the wiki for HamPi 1.2 -- Issue #67

Neon22, 2020-12-06:
#66 and #67 added to wiki

Neon22, 2020-12-06:
Added rel notes and updated to 1.2 in main docs.
I think that's everything...
shall we close ?

dslotter, 2020-12-07:
@Neon22 Yes, that's it. Changes to HamPi 1.2 are frozen, except emergency defects I discover during testing. However, that would not affect the documentation.

Thank you so much for taking the documentation workload off of my shoulders!

closed 2020-12-07 by dslotter

Neon22, 2020-12-07:
You're most welcome dude. Cheers...

Neon22, 2020-12-07:
Hey managed to work out how to get the wallpaper in the docs resized. Front page !

dslotter, 2020-12-07:
Looks good!