dslotter/HamPi #62: Pi 4 8GB won't boot, board requires newer software...

Saebb4366, 2020-11-16:
Hello Prof's,

i have downloaded the newest V2 hampi and flashed it with balena etcher to the 128gb sandisk card, which was with the pi-package. before flashing I formatted the card with sd-card-formatter.

whether I will boot up the pi, I'll get the message on the first boot screen:
'This board requires newer software. get the latest software from:- hits://www.raspberrypi.org/downloads'

boot loader version is 940f978d Mar 19 2020

can you help me, what I have to do to get it run?
[image]

dslotter, 2020-11-16:
Where did you download this from?

Saebb4366, 2020-11-16:
Hei,

I‘ve downloaded it over the torrent link out of w3djs‘ qrz page.

I think the problem is, the 8gb pi4 runs at 64x structure, the actual desktop raspbian just runs on 32x structure (for 1gb - 4gb pi‘s).

I‘will try to copy the missing files (start4.elf etc.) out of the brandnew actual x64 raspbian image and overwrite these of the hampi one‘s.
For sure, with this solution i am running an 32bit system without 64 integration. But for the first solution it hopefully will run on the new 8gb pi just by using 4gb of ram.

But i think for the version 3 development it should be used the newest version for 64bit raspbian. For sure, with this you just run 4gbRam/app, but i think we all know this kind of ram use. 8gb are just for multitasking relevant. What we all do ;-). This is why i run 8gb version, lol.

Go on guys, you do great work! I will test the transplantation these days and will report.

dslotter, 2020-11-16:
Incorrect. I am W3DJS and my QRZ page does not have a download link to HamPi.

It looks like you're trying to boot up the deprecated W3DJS for Ham Radio V2 image I created a year ago. It is several versions out of date. And you're incorrectly calling it HamPi.

Go here and download the current version: http://hampi.radiowaves.ca/

THEN Join the Community Support group at https://groups.io/g/Ham-Pi to get further support.

Closing this as "not a defect".

closed 2020-11-16 by dslotter