dslotter/HamPi #63: Adding Bluetooth settings for audio devices

PA0GTB, 2020-11-19:
Its very difficult bto add a Bluetooth device in HamPi
Its impossible to add a Bluetooth audio device e.g. Speaker to this OS
Its no problem at all to do this in standard Raspbian OS
So there must be a difference in this OS ?
Please take care of it in the new version

keep up the good work !  

73, de Cor, PA0GTB

dslotter, 2020-12-06:
@PA0GTB Cor, HamPi 1.2 will be released with updated Raspberry Pi OS which supplants ALSA with PulseAudio. Please give HamPi 1.2 a try once it's released and see if this corrects your issue.

PA0GTB, 2020-12-09:
Hi Dave

Yes in HamPi 1.2 everything is working OK !!

 

If its OK, I like to ask you a question about restoring old settings from HamPi 1.1 to version 1.2

I think its not working ok

Several ,people here in Holland have problems with it

 

I like to hear from you

 

73 de Cor, PA0GTB

dslotter, 2020-12-09:
Glad to hear Bluetooth is working better in HamPi 1.2.

For support questions, please email Ham-Pi on groups.io.

closed 2020-12-09 by dslotter

PA0GTB, 2020-12-19:
Dave,

Thank you very much for all the good work you do on HamPi

I know that here in Holland its very much appreciated

 

I myself have now HamPi on 2 Raspberry Pi 4 running after new complete installation

I already have send a item to the HamPi Forum on groups.io but No response at all..???

Its about the usage of the possibility of Restoring setting from the prior version 1.1

Itys NOT working. I’ve tried several times and at the the end decided to make a full new installation

On the Dutch Ham Radio forum I now read that also other Hams have the same problem

So there must be something wrong with restoring the old settings

 

Because there is no response on the HamPi forum,, I write you this message directly

Please take care and check the script

 

Thanks

 

73 de Cor, PA0GTB