dslotter/HamPi #71: Step #4 in Build from Source Instructions VERY Vague

JeffHochberg, 2020-12-09:
Step # 4 in the instructions to build from source are incredibly vague:

Populate HamPi/files directory with the SDRplay installer and JTDX (must be manually downloaded) as well as the ham radio wallpapers in HamPi/files/usr/share/rpd-wallpapers.

Which SDRplay installer? There are two files (sdrplay_api2_install.exp and sdrplay_api3_install.exp) that are already in the 'files' directory. There is no installer for SDRplay on the SDRplay website. They have a link to download CubicSDR and SoapySDRplay from third party websites. There's also version 3.07 of the SDRplay API (as of today - 12/08/2020), but "SDRplay installer" is not specific enough.

As far as JTDX is concerned, do we just need to download the .deb package for Raspberry Pi from the JTDX website and drop it into the 'files' directory?

The wallpaper file is already under /files/usr/share/rpd-wallpaper/ (HamPi Desktop Wallpaper.png) so does this satisfy the requirement? Or does it need to be copied directly under the /files folder?

What is the expected behavior if we don't obtain any/all of these files? Will Ansible simply continue on and just ignore the fact that they're missing? Or will it exit the process and the build will fail?

Please add more specifics.

dslotter, 2020-12-09:
@JeffHochberg Good points you raise! I went ahead and updated the wiki page. It specifies the SDRplay installer as "SDRplay_RSP_API-RPi-2.13.1.run" and JTDX is "jtdx-2.2.0-rc152_u_armhf.deb".

I also included the complete list of wallpaper files and the location they go in.

Expected behavior if you don't obtain any or all of these files is that Ansible will terminate in the module referencing the missing files. The process will exit and the build will fail.

dslotter, 2020-12-15:
@JeffHochberg and I are going to run through a build starting from scratch and see if the build instructions are missing any steps that a brand new person would need to know.

closed 2020-12-15 by dslotter