dslotter/HamPi #73: IOWA

johnMXL, 2020-12-23:
Using a Linux Mint 20 base system on Toughbook CF-19F with Intel Core2 Duo U7500.

Downloaded .xz file from link supplied in email.  Verified checksums.  Extracted IOWA.iso using "Extract here" function of file manager.  Used dd from command line to burn .iso file to SanDisk 32gb thumb drive.  Rebooted with thumb drive in USB - PC already set to boot from USB.  Blank screen with flashing cursor.  

Reboot into Linux, view contents of thumb drive.  Thumb drive is blank.

Burnt image with USB stick creator utility in LM20.  View contents.  Thumb drive is blank.

Changed thumb drives and repeated.  Thumb drive is blank.

Downloaded .xz file into different directory.  Repeated extraction, verification, and burn.  Thumb drives are blank.

Viewed IOWA.iso contents using Archive Manager - 2 files: 1.img (16.5Gb) and EFI System Partition.img (536.9Mb).

Did I miss a step somewhere?  What additional info do you need?

dslotter, 2020-12-24:
@johnMXL To attempt to replicate your issue, I downloaded the IOWA.iso.xz beta from the website along with the sha256sum file and confirmed the hash listed in the sha256sum file matched the hash of the xz file. I then ran `unxz -v IOWA.iso.xz` to decompress the archive to its original size.  The decompress operation took 5 minutes, 11 seconds on my Lenovo Thinkpad W550S laptop running Ubuntu Linux. I then copied the ISO to the flash drive by inserting the USB thumb drive into the USB port, confirming the thumb drive was /dev/sdc and then executing:

`dave@Moonquake:~/Downloads$ sudo dd if=IOWA.iso of=/dev/sdc bs=4M conv=fdatasync status=progress`

I observed the activity LED on the flash drive was almost constantly on (some blinking). I also observed that upon completion, the USB flash drive was extremely warm to the touch (almost hot).

dd repoted that it wrote 16GB over 659 seconds at an average write speed of 25.0 MB/s

I am now going to reboot this computer using the USB disk to confirm it starts as expected. I will edit this post once I have completed this task.

dslotter, 2020-12-24:
I am watching the other computer boot HamPi. It is taking a long time, but I saw the Xubuntu logo along with some Linux startup output. It seems to me that writing this ISO using Rufus may in fact allow it to boot faster than writing the ISO using dd. I will note that in the documentation moving forward.

I suggest looking closely at what device you are writing the image to. I suspect that may be where the failure is. How did you determine the correct device to write to? What device are you writing to? Closing this as "not a bug". (But please do follow up and answer my questions, please.)

closed 2020-12-24 by dslotter

johnMXL, 2020-12-25:
To answer your questions:

I determined the drive manually using info from 'lsblk', 'dmesg', File 
Manager, and Disks.

The GUI utilities automatically selected the flash drive.

On this system the flash drive appears as /dev/sdb - and when it was 
recognizable would be auto-mounted to 
/media/johnmxl/_Long_Device_ID_Number_.

To update:

I did a fresh download today and used Rufus on Win10 to flash the image 
to the same 32GB SanDisk Cruiser I've been using all along, and got the 
same result attempting to boot the Toughbook.

I took the drive to 2 desktop PCs - one Dell, one HP, and both booted 
into HamPC.

I'm going to pick up a couple of new thumb drives this weekend - stores 
will be closed tomorrow for some reason - this is the only 32GB I have 
free ATM - all the rest are too small - and will try again.

I've not had problems burning and booting other live USB images on this 
drive and this laptop - don't know why IOWA would be any different.

John K0MXL

On 12/24/20 10:02 AM, Dave Slotter wrote:
> I am watching the other computer boot HamPi. It is taking a long time,
> but I saw the Xubuntu logo along with some Linux startup output. It
> seems to me that writing this ISO using Rufus may in fact [...]

dslotter, 2020-12-25:
Excellent detective work!

Can you please provide the model of ToughBook you are having the issue with?