dslotter/HamPi #74: IOWA - alpha

w0kah, 2020-12-25:
Hamradio - Fldigi Suite
fldigi terminates without showing opening config dialog
ditto flamp, flcluster, fllog, flmsg, flrig, flwkey

flarq terminates w/msg fldigi not running

flnet shows opening dilog, ok'ing default filename terminates, same if a filename is entered
debug log was created, contents:
I: main: flnet debug log
I: main: Opened: /home/hampc/flnet.files/new_net.csv
the above file was NOT created

flpost opens, have not used so did not go further on this app

flwrap works dragging & dropping file

PC: HP 400-224
AMD A4-5000 apu
12 gB memory
samsung ssd 500 gB
complete specs without the ssd & memory upgrades:
https://support.hp.com/us-en/document/c04078971

wired 100 mb ethernet
HP w2338h monitor - had to autoscan to eliminate desktop from being overscanned slightly a few pixels on all sides; seconds cut off on time, "home hampi icon" really scrunched up into top left corner" 

Spot checked a few other apps and they seem to work, ham radio emcomm is my main focus.

Ken-W0KAH

dslotter, 2020-12-25:
I'm seeing fldigi start and show the initial config dialog, but it is frozen -- e.g. not responding to mouse clicks. Launching it from the terminal, this is the output:

hampc@hampc:~$ fldigi  
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card

dslotter, 2020-12-25:
Flamp starts up from the command line with no output to stdout, and displays a dialog saying, "Start fldigi before flamp!" and then hangs.

dslotter, 2020-12-26:
Hi @w0kah . After restarting the window manager -- which locked up on an unsupported platform -- I was able to successfully launch flrig, fldigi, flamp, flcluster, fllog, flmsg, flnet, flwkey. I performed initial setup on all requesting that, and was able to successfully shut each down.

So just because I was able to do this without error doesn't excuse what happened to you. I believe you had these difficulties, and the only thing to try to figure out is -- why?

Please give me some time to look into this. This may not be a quick fix -- especially since I do not have access to a HP computer.

What you could do for me is to provide the following files if they exist:

$HOME/.fldigi/debug/debug_log.txt (or any files in this directory)
$HOME/.flcluster/debug_log.txt
$HOME/.flrig/debug_log.txt
$HOME/flnet.files/flnet_debug_log.txt
$HOME/.nbems/debug_log_.txt
$HOME/.nbems/FLAMP/debug_log.txt

dslotter, 2020-12-26:
@w0kah Do you have a second computer you can try Project IOWA against?

w0kah, 2020-12-26:
I have others I can try on.  Will look into it more tomorrow when I get
home from work. Probably 10 AM or so ct.
73, Ken W0KAH

w0kah, 2020-12-26:
Dave,

Seems IOWA is uefi boot, two of my other "test" systems are legacy boot, so
I'm giving it a try on my HP Elitebook 840G3. I normally try to keep that
restricted to Microsoft Windows 10 preview edition activities. However, I
did get IOWA to boot up after a little finagling to bypass some security
features.

That said, fldigi suite seems fine on this computer, which blows my mind.
The HP 400-224 that I was using ran Windows 10 Pro and all the windows apps
I threw at it without issues. It also runs Fedora 33 just fine. In fact I
used it to download IOWA and uncompress it...

re: IOWA, one thing I wasn't expecting is that flxxx working files seem to
be in /home/hampc/.fldigi/ which is hidden in the default view of the
"home" file system when I use the desktop icon. I looked in fldigi under
folders to see where it was depositing "stuff." Then I saw lots, so might
make a note to let others know to enable view hidden files in file
explorer...

Again, all seems fine on this laptop.

73 and keep up the great work on HamPi!

Ken - W0KAH

dslotter, 2020-12-26:
Hi Ken,

Yes, IOWA is UEFI boot, although if I have some time, I might create a
legacy boot version.

I've just about finished the Ansible part for creating HamPC 1.2a_rc1 (yes,
you guessed the final name but please don't give it out early). I fixed the
issues that you and another person brought up... D-Rats and HamClock both
work now, and I got Chirp to install as well as PyQSSTV. I expect to post
this later today and make available for general download as a "release
candidate".

-Dave