dslotter/HamPi #75: Computer Locks-Up on HamPC Shutdown

Ken-K1KDA, 2020-12-25:
When choosing Shutdown from the menu, the computer begins a normal shutdown and then locks up.  The only way to clear the lockup is to use the computer main power button to power down for a reboot.

dslotter, 2020-12-26:
@Ken-K1KDA I tried the same on my Intel NUC (unsupported platform for graphics) as well as Dell Latitude, and both shut down the computer completely. Something is different about your computer architecture. We will have to dig deep and investigate.

Ken-K1KDA, 2020-12-26:
Dave,

The computer is not homebrew but it is an older model.  I suspect the issue is related to IO overload.  It is my main shack computer and controls 3 Icom Radios, a AZ rotor and 2 displays.  

I re-ran the test again on a different PC and IOWA shutdown gracefully without issue.  So, I am going to put this issue aside for now and switch to a different test environment.  My next test environment will be a 16GB Windows Pro machine with IOWA running simultaneously in an Oracle Virtual Memory (VM) Box which is my preferred method of Linux operation.  It will take me a couple of days to setup the new environment.

dslotter, 2020-12-26:
Ken,

Thanks to your feedback, the next release of IOWA will have D-Rats and
HamClock now fully operational, along with Chirp being installed (having
benefited from the investigation/changes for D-Rats.)

Your reports are much appreciated!

73,

-Dave