dslotter/HamPi #76: HamPC Programs Failed to Start

Ken-K1KDA, 2020-12-25:
Ran a simple start or no-start test of all the applications listed in the main menu under Hamradio.  The following applications failed to start without any indication as to why.  No error codes just no response.

- NE2C
- D-Rats
- D-Rats repeater
- FLRig/FLDigi and associated programs
- CubicSDR
- Gqrx
- Quisk
- SDRAngel
- FreeDV
- HamClock
- TwClock

Test system configuration:
Windows 10 Home (x64)
2.10 gigahertz AMD A6-3600 APU with Radeon HD Graphics
Multi-core (4 total)
8178 Megabytes Usable Installed Memory
Sony 500GB SATA SSD
NVIDIA GeForce GT 610 Display adapter

-

dslotter, 2020-12-26:
I ran this same test on my Intel NUC:

### Started up successfully
FLRig/FLDigi and associated programs
CubicSDR - Used SDRPlay RSP2 and it ran as expected
Gqrx
Quisk
SDRAngel
FreeDV
TwClock

### Failed to start
NE2C
D-Rats
D-Rats repeater
HamClock

D-Rats is almost definitely missing a Python 2 module. I had difficulties with this when building the image. I will try my hardest to get this working, because many folks use D-Rats for EMComm. NE2C brought up a dialog for 1/10 second and then disappeared before I could read it, and HamClock didn't display anything at all. I will look into all these.

dslotter, 2020-12-26:
HamClock is a good find -- this is happening because passwordless sudo isn't working. I'll correct this.

dslotter, 2020-12-26:
NEC2C is a really good find -- this is a command-line tool shoehorned into a GUI menu and hasn't worked since it was put into HamPi. I'll prioritize to fix this after D-Rats.

dslotter, 2020-12-26:
D-Rats has had its missing libraries added for Xubuntu.

Ken-K1KDA, 2021-01-02:
Dave,

I have been "under the weather" for a couple of days.  I hope to get back
on track in a few more days.  I apologize for not completing more IOWA
testing.  I will get back to it as soon as I can.

Ken
K1KDA