dslotter/HamPi #79: Unable to burn image

jlauber, 2020-12-31:
Dave--I've followed your instructions re downloading the image file (including checking SHA256 data) multiple times, am running the latest version of Rufus and am using virgin 32GB usb sticks (tried two), and cannot get Rufus to write the image.  I can select the proper Device ok and I can select the image from my Downloads folder ok (HamPC_v1.2b_rc1.iso.xz) and from there nothing happens.  Every other window is greyed out except  START.  If I hit START, Rufus first checks for "conflicting processes" then indicates "Deleting Partitions (this may take awhile)" and after a few seconds, indicates "writing image 0.1%..."  After less than two seconds, Ready turns green and that's the end of the process.  I get identical results when trying this on two of my Win 10 machines (HP and a Surface Pro4).  I can use Rufus to write a FreeDOS image if I select that, but am unable to make it write the HamPC image.  What am I doing wrong??  Tnx, John WI6P

encbar5, 2021-01-03:
Perhaps you need to decompress the xz archive first. Then Rufus should work fine on the iso file.

dslotter, 2021-01-27:
@jlauber I know that Rufus can write the HamPC image, but perhaps something is unique to your situation that is preventing you from writing it. That said, I'm currently trying to repackage the HamPC image into a bootable ISO, instead of the bootable image it currently is. I'll let you know when I've repackaged the HamPC image and you can try again then...