dslotter/HamPi #81: alsa soundcard error

hans-dot, 2021-01-05:
Sound not working.  syslog errors:
Jan  5 05:47:39 hampc systemd-udevd[323]: controlC0: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore 0' failed with exit code 99.
Jan  5 05:47:39 hampc alsactl[728]: alsa-lib main.c:1021:(snd_use_case_mgr_open) error: failed to import hw:0 (empty configuration)
Jan  5 05:47:39 hampc alsactl[728]: Found hardware: "HDA-Intel" "Realtek ALC892" "HDA:10ec0892,18498892,00100302 HDA:80862806,80860101,00100000" "0x1849" "0x8892"
Jan  5 05:47:39 hampc alsactl[728]: Hardware is initialized using a generic method
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #3 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #4 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #6 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #7 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #8 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #9 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #17 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #18 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #19 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #27 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #28 (No such file or directory)
Jan  5 05:47:39 hampc alsactl[728]: /usr/sbin/alsactl: set_control:1325: failed to obtain info for control #29 (No such file or directory)

dslotter, 2021-01-05:
@hans-dot Thank you for providing this information. What brand and model of PC are you running? Can you provide any hardware specs?

dslotter, 2021-03-07:
Closing due to lack of response after two months.

closed 2021-03-07 by dslotter