dslotter/HamPi #149: Dump1090 fails to link

dslotter, 2021-11-27:
Not sure if this is a new bug or a Debian Bullseye issue yet. Needs investigation.

dslotter, 2021-11-27:
Not familiar with this "multiple definition" linking issue. Somehow we're getting two copies of functions in the link process.

`gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o demod_2000.o demod_2400.o demod_8000.o stats.o cpr.o icao_filter.o track.o util.o convert.o dump1090.h -L./ -lsdrplay_api -lpthread -lm -lrtlsdr  -lrt `pkg-config --libs librtlsdr`  
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:403: multiple definition of `Modes'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:403: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:411: multiple definition of `slaveAttached'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:411: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:410: multiple definition of `slaveUninitialised'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:410: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:410: multiple definition of `masterInitialised'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:410: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:408: multiple definition of `chParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:408: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:407: multiple definition of `deviceParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:407: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:406: multiple definition of `cbFns'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:406: first defined here
/usr/bin/ld: interactive.o: in function `.LANCHOR0':
/home/pi/hamradio/dump1090/dump1090.h:405: multiple definition of `chosenDev'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:405: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:411: multiple definition of `slaveAttached'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:411: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:410: multiple definition of `slaveUninitialised'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:410: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:410: multiple definition of `masterInitialised'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:410: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:408: multiple definition of `chParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:408: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:407: multiple definition of `deviceParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:407: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:406: multiple definition of `cbFns'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:406: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:405: multiple definition of `chosenDev'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:405: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/hamradio/dump1090/dump1090.h:403: multiple definition of `Modes'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:403: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:403: multiple definition of `Modes'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:403: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:411: multiple definition of `slaveAttached'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:411: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:410: multiple definition of `slaveUninitialised'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:410: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:410: multiple definition of `masterInitialised'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:410: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:408: multiple definition of `chParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:408: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:407: multiple definition of `deviceParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:407: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:406: multiple definition of `cbFns'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:406: first defined here
/usr/bin/ld: mode_s.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:405: multiple definition of `chosenDev'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:405: first defined here
/usr/bin/ld: net_io.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:403: multiple definition of `Modes'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:403: first defined here
/usr/bin/ld: net_io.o: in function `.LANCHOR1':
/home/pi/hamradio/dump1090/dump1090.h:411: multiple definition of `slaveAttached'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:411: first defined here
[... cut ...]
/usr/bin/ld: convert.o:/home/pi/hamradio/dump1090/dump1090.h:407: multiple definition of `deviceParams'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:407: first defined here
/usr/bin/ld: convert.o:/home/pi/hamradio/dump1090/dump1090.h:406: multiple definition of `cbFns'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:406: first defined here
/usr/bin/ld: convert.o:/home/pi/hamradio/dump1090/dump1090.h:405: multiple definition of `chosenDev'; dump1090.o:/home/pi/hamradio/dump1090/dump1090.h:405: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:54: dump1090] Error 1
`

Mopar44084, 2022-01-09:
What is it doing? I figured out a fix so that it displays aircraft in the browser!

dslotter, 2022-01-11:
This was fixed in the HamPi 2.0 release.

closed 2022-01-11 by dslotter