dslotter/HamPi #170: Add hfterm

dslotter, 2022-02-01:
It was mentioned that Linux options for decoding RTTY were limited to "fldigi, linpsk, hfterm and gMFSK", so adding hfterm.

dslotter, 2022-05-08:
hfterm may be found at http://hfterm.sourceforge.net/

dslotter, 2022-05-08:
Cannot build on modern GTK2 or GTK3 system without major rework of configure and Makefile, so this will not be included with HamPi. Filed issue at https://github.com/DavidGriffith/hf/issues/1

closed 2022-05-08 by dslotter

DavidGriffith, 2022-05-29:
FYI, I'm not a maintainer of hfterm.  I just tossed it up on Github to tinker with it and hopefully fix its problems.  Upstream has version 0.9 whereas my repo has 0.8.

DavidGriffith, 2022-06-12:
If anyone knows how to get GTK1 installed on a modern Debian system without much trouble, I'd love to hear from you.  With that, I could possibly port hfterm to GTK2 or GTK3.