dslotter/HamPi #172: Version 2.0 problems connecting to Icom IC7300 with Vfview

tegwilym, 2022-02-08:
I'm trying the HamPi 2.0 version and moved all my settings over.  When trying to connect to my IC7300,  nothing happens.  On the older version (1.1 or something?) it all works fine when set to "automatic" for the usb port setting.  Nothing  on the 2.0 version works.   It is Vfview 1.2d version.  I've posed there too and they say the port isn't owned, so I need to mess with that a bit.  Or maybe when I moved all my settings over there was something not compatible?
Just wondering of anyone else has has this issue?

tegwilym, 2022-02-09:
Some more info:  I made a fresh copy of HamPi from the image and tried it as a fresh copy with nothing else done.  Wfview worked fine "out of the box" and connected to the 7300 fine.  Then when I did the 'apt update / apt upgrade' and rebooted, I was back to where I was - no connection from Wview and 7300 again. The forum guys on the Wfview side, say that it may be something not releasing the port or a permissions problem.  Just not sure yet what it is.  I just go back to the older HamPi and it's fine, I need to try comparing connections on that one maybe.

geezrgeek, 2022-02-18:
/dev/serial is not showing up ub ver 2.0b1, I think the serial driver has been moved to a module that is not being loaded by HamPI.  If I knew what module it is now in, I would install it.

dslotter, 2022-05-08:
@tegwilym I think that the ORCA screen reader software may be interfering with the port. I'm removing it from the next release of HamPi because it's caused more problems than it solved.

closed 2022-05-08 by dslotter