dslotter/HamPi #175: HAMPI README.docx to PDF?

LarryGFlorence, 2022-05-14:
Change format of HamPi/files/home/pi/Desktop/HAMPI README.docx to PDF?

dslotter, 2022-05-14:
What's the advantage of doing this?

LarryGFlorence, 2022-05-14:
Theoretically, it comes down to using the right tool for the job. As I understand it, .docx is a document _editing_ format, while .pdf is a document _distribution_ format. Furthermore, .docx is proprietary and depends on Microsoft software for the "fidelity" of display; .pdf is an open standard intended for full "fidelity" across platforms and installed fonts, etc.

Practically, on a "virgin" copy of HamPi 2.0b1, double-clicking on the .docx file opens in "xarchiver" leading nowhere; there is no application on board for that format. The "GIVE BACK.PDF" document opens in Chrome. The "HAMPI LICENSE.TXT" file opens in the Geany IDE (?).

LarryGFlorence, 2022-05-17:
Now that LibreOffice is part of the HamPi 3.0 application set, the HAMPI README.docx file opens. But it is not a pretty sight and the hyperlinks are not directly clickable.

Consider exporting the document from LibreOffice as a PDF and distributing that instead. Shown below is what the README file looks like before and after.

(PS. Same improvement could be made to HAMPI LICENSE.TXT)

[HAMPI README.pdf](https://github.com/dslotter/HamPi/files/8704846/HAMPI.README.pdf)

[image]
[image]

dslotter, 2022-05-17:
I see where you're coming from.

However, converting to a PDF every time would increase my burden and build time, and considering the end user is capable of converting to PDF, I will leave it in the end user's capable hands.