dslotter/HamPi #176: Problems with /home/pi/bin/update_fldigi_suite

LarryGFlorence, 2022-05-14:
Running "Debian GNU/Linux 11 (bullseye)"

1. The test for Ubuntu causes the wrong libusb to be installed, resulting in an error.
2. Multiple downloads of a tar.gz file result in a failure to extract the archive.
3. Cleaned up some status messages.

[update_fldigi_suite_diff.txt](https://github.com/dslotter/HamPi/files/8691862/update_fldigi_suite_diff.txt)

31,42d30
< ubuntu_version=`lsb_release -sr | awk -F \. {'print $1'}`
< if [ $ubuntu_version -lt 20 ]; then
< {
<     echo "Ubuntu pre-version 20 detected..."
<     # Ubuntu earlier than 20.04:
<     sudo apt -y install build-essential libfltk1.3-dev libsamplerate0-dev portaudio19-dev libsndfile1-dev libxft-dev libxinerama-dev libxcursor-dev libpulse-dev pavucontrol libusb-1.0.0-dev
< }
< else
< {
<     echo "Ubuntu version 20 or later detected..."
<     # Ubuntu earlier than 20.04:
<     # Ubuntu 20.04:
44,45d31
< }
< fi
51c37
<   echo "Installed ${PROGRAM} version is: $INSTALLED_VERSION"
---
>   echo "Installed ${PROGRAM} version is: ${INSTALLED_VERSION}."
64c50
<   echo "Installed ${PROGRAM} version is ${CURRENT_VERSION}."
---
>   echo "Current ${PROGRAM} version is: ${CURRENT_VERSION}."
71c57
<   if ! wget https://sourceforge.net/projects/fldigi/files/${PROGRAM}/${PROGRAM}-${CURRENT_VERSION}.tar.gz; then
---
>   if [ ! -f ${PROGRAM}-${CURRENT_VERSION}.tar.gz ] && ! wget https://sourceforge.net/projects/fldigi/files/${PROGRAM}/${PROGRAM}-${CURRENT_VERSION}.tar.gz; then

dslotter, 2022-05-14:
Thank you for providing a diff. I'll try to see if I can make this disconnected-looking piece of code work with the patch program.

For future reference, this script can be found at: https://github.com/dslotter/ham_radio_scripts/blob/main/update_fldigi_suite and you can send a PR (pull request) which would make it easier for me to integrate the fixes.

dslotter, 2022-05-14:
Nevermind about the disconnected comment. I found the diff file.

dslotter, 2022-05-14:
Fixed in git commit 08a241b (in ham_radio_scripts) repository.

closed 2022-05-14 by dslotter