dslotter/HamPi #177: Add version string to HamPi Desktop Image

LarryGFlorence, 2022-05-14:
Perhaps add a version string to a corner of the HamPi desktop image?

dslotter, 2022-05-14:
There is a version string embedded in the file system. Please see the file: /etc/hampi-release

`pi@hampi:~ $ cat /etc/hampi-release
PRETTY_NAME="HamPi 2.0 (December 2021)"
NAME="HamPi"
VERSION_ID="2,0"
VERSION="2.0 (December 2021)"
HOME_URL="http://hampi.radiowaves.ca"
SUPPORT_URL="https://groups.io/g/Ham-Pi"
BUG_REPORT_URL="https://github.com/dslotter/HamPi/issues/"
`
Also, there is a version check which happens during startup and alerts to running an older HamPi version than what is released.

closed 2022-05-14 by dslotter

LarryGFlorence, 2022-05-14:
The version check script is mentioned in /home/pi/.config/lxsession/LXDE-pi/autostart, but the script is missing (HamPi 2.0b1)?

Suggestion: Add an "About HamPi…" item to the Ham Radio menu, to make the version information more accessible:

/usr/share/icons/about_HamPi.png
/usr/share/applications/about_HamPi.desktop
/usr/local/share/about_HamPi.sh

See the attached tar file (which was zipped so that it could be attached to this issue).

[about_HamPi.zip](https://github.com/dslotter/HamPi/files/8694005/about_HamPi.zip)

dslotter, 2022-05-15:
Added in git commit e174515