dslotter/HamPi #179: Initial Raspberry Pi OS Username/Password Entries Are Not Saved

LarryGFlorence, 2022-05-17:
The first time a new HamPi 3.0 installation runs, a dialog asks for a new User Name and Password. The entries of this dialog are not saved; neither the user name nor the password are set. There is no error message.

This results in the confusing warning about SSH using the default pi password. It also leads to confusion when the user needs to enter a password, for example, for sudo execution.

LarryGFlorence, 2022-05-18:
Furthermore, I found that the settings made in Imager pre-burn *are* made in the resulting distribution - user name, password, WiFi settings, etc. But, oddly, the QSLWare dialog does not appear :(

Loonhaunt, 2024-04-07:
I had been using ham pi without changing the password for a short time, it now is asking for a password to perform gpsd , in getting a Ublox7 to talk and it is asking for a password?  None of the known rpi intial passwords are working?  Any suggestions other than burning a new image?

Dave
KJ5FKI