dslotter/HamPi #182: Enabling WLAN AP mode possible?

rmx-it, 2022-05-22:
Hi,
would it be possible to activate the RPi in WLAN Access-Point mode so that one could
use the RPi headless via VNC with mobile phone as client?
Could ease portable usage...

dslotter, 2022-05-22:
Would it be possible? Yes. However, I tried this over two years ago and while it was great for some people for the majority it caused a lot of problems, so I went ahead and removed it. Please let me know if you need instructions on how to set this up, and I will point you to a good web page with instructions.

closed 2022-05-22 by dslotter

rmx-it, 2022-05-22:
Hi,
yes a pointer would be really helpful.

dslotter, 2022-05-26:
Some links for you @rmx-it :

https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/
https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point/install-software
https://www.raspberrypi.com/documentation/computers/configuration.html
https://raspap.com/