dslotter/HamPi #183: V3 USB Boot failure

Neon22, 2022-06-03:
I have a pI400 unit.
I have updated the eeproms by booting into a new fresh os and using piconfig to set boot order to usb then sdcard. I have verified I can boot into a new USB distro. so this part seems to be working.

on booting into the hampi v3 new release - i get an error
"failed to fixup PID1 environment" 
Various searches lead me to believe its something to do with systemd vs init.d setups but its a mystery to me.

Any idea on what change I need to make to enable boot from USB ?

dslotter, 2022-06-03:
I'm actually not as familiar with USB boot as I would like, so I've crossed posted this to the HamPi community support forum on groups.io so that somebody else may be able to answer this.

magar832, 2022-08-02:
I have found a solution to allow HamPi to boot. Currently testing this on v3.0.1 64-bit for the Pi 400. 

I have enabled USB boot using the official Raspberry Pi Imager, **_however I use microSD cards for HamPi and RaspiOS_**. 

Official Raspberry Pi OS boots normally. HamPi does not, and the Pi 400 will boot loop. 

### Solution: 
Commenting out _either_ of these two lines in the HamPi's **config.txt** in the boot partition allows HamPi to boot normally. 
- `hdmi_safe=1`
- `hdmi_force_hotplug=1`

HamPi should boot up after commenting either or both of these out. Restoring these values to their uncommented state still allows HamPi to boot up. So, something about these values, or perhaps the config file as a whole, allows HamPi to boot normally after its first boot.