dslotter/HamPi #205: HamPC v3.0.1b1 Image Won't Burn to Thumb Drive

NV4C, 2022-09-04:
I downloaded the file HamPC_v3.0.1b1.img.xz to my Windows 10 PC. I followed the instructions in the HOW_TO_FLASH_IMAGE.TXT file, to use Rufus to extract the ISO to my 128 Gb thumb drive. Rufus takes about 5-10 seconds to run through its process. When complete, the thumb drive is empty and is now undetectable by my Win10 PC. When I insert the drive in my Linux Mint laptop, it is detected, but shows as empty.

The thumb drive is an SP Blaze B02 USB 3.2 Gen 1 Flash drive obtained from Amazon.

Here is a screen shot of Rufus, before I hit "Start". Is there something else I need to select to properly burn the ISO to the thumb drive:

[image]

Thank you for your assistance.

Ian Kahn, NV4C
nv4c.ian@gmail.com

dslotter, 2022-09-09:
Rufus appears to have some issues with regards to reliability. I'm going to change the instructions to point to Balena Etcher instead.

Neon22, 2022-09-21:
I used the Raspberry Pi Imager with great success.