dslotter/HamPi #220: Won't start the desktop on  Pi3B+

neil969, 2023-04-04:
**Describe the bug**
I have burnt V3.3 32 bit on an SD card a few time ( using Balena & the Pi Imager) but it fails on first boot 

**To Reproduce**
Steps to reproduce the behavior:
1 Burn SD
2 stick it in ( attached to an HDMI Monitor
3 power it up
4 See it error :) 

**Expected behavior**
Boot to a desktop

**Screenshots**
Can't sorry 
But it shows  the  pi boot screen and then goes black
then [Failed} Failed to start Remount Root & Kernal File Systems
[Failed} Failed to start SuxLink remote tranceiver repeater control software
[Failed} Failed to startSuxLink repeater control software
[Failed} Failed to start MariaDB 10.5.15 databse server

The the usual terminal start blurb up to Use raspi-config to set the country before use. 
Then a prompt.  

**System Info (please complete the following information):**
 - OS: HamPi 
 - HamPi or HamPC Version 3.3
 - For HamPi, pi3B+
 -

**Additional context**
Add any other context about the problem here.

ka3yan, 2023-04-04:
I'm having a similar issue with a RPi 4b.

[FAILED] Failed to start Remount Root and Kernel File Systems.
[FAILED] Failed to start Check for Raspberry Pi EEPROM updates.
[FAILED] Failed to start MariaDB 10.5.15 database server.
[FAILED] Failed to start SvxLink remote transceiver repeater control software.
[FAILED] Failed to start SvxLink repeater control software.

Same as above.

I did reformat the SD card between each imaging attempt.

dslotter, 2023-04-04:
I'm investigating these reports.

N1MCJ, 2023-04-07:
I am having the same issues.
[image]

msamis1, 2023-04-10:
I have the same issue with exactly the same errors with a RPi 4B.  I tried to update in command line but wifi is "blocked by rfkill" I then went down to ver. 3.01 and it loaded up with no problems.

IZ5RZR, 2023-04-15:
Hi. Same problem on my PI3, reflashes more time but no WORK.
Where can we download previus versions?

msamis1, 2023-04-15:
https://sourceforge.net/projects/hampi/files/
I used [2022-09 Release](https://sourceforge.net/projects/hampi/files/2022-09%20Release/)

IZ5RZR, 2023-04-16:
Grazie!

   Ciao
   Da "msamis1" ***@***.***
   A "dslotter/HamPi" ***@***.***
   Cc "Max" ***@***.***,"Comment" ***@***.***
   Data Sat, 15 Apr 2023 12:30:04 -0700
   Oggetto Re: [dslotter/HamPi] Won't start the desktop on Pi3B+ (Issue #220)

   https://sourceforge.net/projects/hampi/files/
   I used 2022-09 Release

Granmo, 2023-04-21:
[image]

Same issue I have done numerous burns to the same result

Thanks for any help you can give

Regards

Steve

msamis1, 2023-04-21:
Have you tried running raspi-config to set up the location?
Have you tried to go down to a lower version?  Are you using a RPi 4B? I had to do that with an RPi 4B with 4gb. If you have a lower version or less memory, you may need to keep going down to older versions until it works.

IZ5RZR, 2023-04-21:
Hi guys
We can solve iussue only flashig previous version. IT WORKS!

Granmo, 2023-04-21:
I have a raspberry pi 4 b with 2 gig. I can use an older version but it will not run wsjtx, so not for me. I will wait until they figure a fix.

Thanks for reply

Regards,

Steve

       Stephen  Granmo
        Principal Broker
            Licensed in OR & WA
C: 503.789.8341   F: 503.200.1096
       www.nwservicerealty.com <http://www.nwservicerealty.com/>
          
      NW ★
      Service Realty, LLC

filipjonckers, 2023-07-26:
Issue confirmed here - RPI 3B+ with brand new 64GB high quality SD card
Did not start in graphical mode and after first boot refuses to enable WiFi - giving the dreaded: Wi-Fi is currently blocked by rfkill error
Maybe best to remove the image from Sourceforge until fixed.
v3.2 is working fine

Linked to #224