dslotter/HamPi #224: Hampi V3.3 32 bit will not run

Harry-1167, 2023-05-03:
Hi, I have downloaded the image twice and installed on 64gb sd card 3 times all was successful but the program will not run it shows some errors which i can't remember at moment but will open up terminal mode which functions.
question is what am i doing wrong all other versions worked ok
I also found a short V3.3 program that looks like a checksum how does it work.
cheers Harry

rag2, 2023-06-07:
There is a problem with the partition for the **rootfs** filesystem on the V3.3 image. The filesystem is around 19 GB, but the partition has been set to 120 GB (111.54 GiB). Thus:
a) any medium smaller than 128 GB (nominal) will fail to install HamPi
b) even after adjustment, the rootfs is **not** auto-adjusted to fill the available space
c) something is clearly amiss at the tail end (or perhaps beginning) of the build process.

There is a fuller description and discussion of this in a short report, which is available at [https://groups.io/g/linuxham/files/Note%20on%20HamPi%20V3.3.pdf](url) if you have access to the groups.io linuxham group.

filipjonckers, 2023-07-26:
Issue confirmed here - RPI 3B+ with brand new 64GB high quality SD card
Did not start in graphical mode and after first boot refuses to enable WiFi - giving the dreaded: Wi-Fi is currently blocked by rfkill error
Maybe best to remove the image from Sourceforge until fixed.
v3.2 is working fine

filipjonckers, 2023-07-26:
Linked to #220

Harry-1167, 2023-07-28:
Hi All, I fixed the problem on my raspberry pi 4b 8gb Hampi 3.3 using 
gparted with the help of Robin G8DQX see below copy of email below hope 
it helps cheers Harry VK2DWT

    Hello Robin, Well i am using ubuntu 23.04 and my hp laptop has a
    card reader so placed the 64gb sd in and ran gparted amazing how the
    rootfs was 111 gb anyway took a little bit of head scratching as I
    have only looked at it once before, so finally figured out how to
    apply your suggestions resized and now runs 100% fine thanks so much
    i learnt a bit with your help and others too, be interesting to see
    what else you came up with wen you get the time thanks again all
    cheers Harry 73 vk2dwt

    On 6/3/23 10:27, G8DQX wrote:

        Harry,

        I've done some digging, in amongst various other duties. The
        long & the short of it is:

            1. You can successfully install HamPI V3.3 to a 128 GB
            medium (medium = microSD card or USB flash storage device)

            2. Installation to a 32 GB or 64 GB medium requires a *hack*
            to make it work

            3. That hack works on a Linux box, such as your Raspberry
            Pi, if one has a *USB card reader*

                a) boot up the Pi with standard Pi OS (minimal desktop)
                (16 GB is more than enough medium)

                b) install *GParted*

                c) connect the USB card reader, and insert the HamPi
                card in it

                d) bring up GParted

                e) every time it complains about excess size, click the
                *Ignore* box. [Do *not* select cancel!] This will happen
                quite frequently at various later stages of this procedure

                f) select the HamPi medium (*probably /dev/sdb*), locate
                the *rootfs* partition. Note that the size of the
                partition is somewhat greater than the size of the
                medium in use. This is not good, and is the direct
                source of the failure to boot

                g) make sure that the partition is not shown as mounted.
                Then *shrink* the rootfs partition to the smallest size
                (that smallest size will be indicated in the dialog box)

                h) and now *expand *the rootfs partition to the largest
                size (GParted will set that largest size to that which
                the card correctly allows)

                i) one should now, with a following wind, have a
                bootable HamPi medium, using all available storage!

        There is more, which I hope to put out on Saturday (my time, not
        yours, and no guarantees.)

        73, Stay Safe,

        Robin, G8DQX

--------------sXlyvAV3WecRaKaRN8sXVeT2
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit
[... 79 lines cut ...]
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Linked to <a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="1653952851" data-permission-text="Title is private" data-url="https://github.com/dslotter/HamPi/issues/220" data-hovercard-type="issue" data-hovercard-url="/dslotter/HamPi/issues/220/hovercard" href="https://github.com/dslotter/HamPi/issues/220" originalsrc="https://github.com/dslotter/HamPi/issues/220" shash="n/sL+NYWUNHNJrIJV7L5+hanGcoe3Px+eCiCEz+0kMlqBau8dh+gA8YmAK8y/agjpAF94Y9F6VppR3wB/1ZpHjDk/cnsLNqY0I+M8XQ2GJlpE6FjXvxGvHjo7+YXc5dzSfeju38yKSjXV7Mp25dAJ47jnmuYmalublTGTtCbwT8=" moz-do-not-send="true">#220</a></p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          visibility: hidden; overflow: hidden; opacity: 0; width: 0;
          height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;dslotter/HamPi/issues/224/1651848539</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/dslotter/HamPi/issues/224#issuecomment-1651848539",
"url": "https://github.com/dslotter/HamPi/issues/224#issuecomment-1651848539",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------sXlyvAV3WecRaKaRN8sXVeT2--

recycled-again, 2024-01-21:
Thank you Harry-1167 for supplying the solution to this frustrating problem. I am surprised, however, that this problem has not been addressed yet or not even flagged on the download page - I can't imagine that many people have used 128GB SD cards!