dslotter/HamPi #225: FRN client

sv2snh, 2023-06-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
At the next update please put in the FRN CLIENT.

dslotter, 2023-06-19:
What is the FRN client and more importantly where do I find it? URL please.

sv2snh, 2023-06-19:
-- ------------------------------------------------------------------------------------------
                    SV2SNH  <-----> HELLAS-FRN------ΕΝΗΜΕΡΩΣΗ 28-05-2023
---------------------------------------------------------------------------------------------
                           Σε ολο το δικτυο το ιδιο TG:20222

  Παγκόσμιο αναλογικό δίκτυο μέ τή χρήση του υπάρχοντος εξοπλισμού πού μπορεί νά έχει κάποιος
  ραδιοερασιτέχνης.
  Συμμετέχουν αρκετοί απο διαφορες τοποθεσίες τής Ελλάδος αλλά καί εκτός αυτής.
  Και προσκαλουμε και αλλους.Ελευθερη χρηση.
  Εφαρμογή στό κινητό {GRN},απο το αυτοκινητο [GRN DROIDSTAR] από υπολογιστή {CLIENT} καί από  πομποδεκτη.RF,
  

 ------------------------------------------------------------------------------------------------ 
 | Το δικτυο σε ολα τα ψηφιακα DMR-Brandmeister,NXDN,P25,D-STAR-XLX165,TGIF,ASL,YSF.M17-FRN....
 | HELLAS-FRN   AΝΑΛΟΓΙΚΟ     Δωμάτιο Ham1                                                      |
 | Βrandmeister 2021 TG20222  Δωμάτιο Ham2                                                      |
 | HELLAS-FRN   HF 29.250Mhz  Δωμάτιο Ham3                                                      | 
 | FreeDMR Greece    TG20222  Δωμάτιο Ham4                                                      |
 | ΥΠΟ ΚΑΤΑΣΚΕΥΗ     TG       Δωμάτιο Ham5                                                      |                                                                                      
 ------------------------------------------------------------------------------------------------
     
     XLX Peer Interlink Settings
     -----------------------------
  Please contact: SV2SNH for XLX peering
  ***@***.***

  Note: Currently only Module A is used for International interlinks
  Other Modules can be done on request.

  Edit xlxd.interlink, add the following line:
  # xlxd.interlink
  XLX165 xlx165.ddns.net A
 -------------------------------------------------------------------------
  
  SV2SNH:QRZ  https://www.qrz.com/db/SV2SNH
  Kόμβος θεσσαλονίκης:145.2875 T 74,4 Ham1 & echolink zello rf  Ham1
  Κόμβος θεσσαλονίκης¨434.0250 Τ 71.9 Ham2 & echolink rf & digital  Ham2
  Κόμβος Χαλκιδικής  :145.2375 T 71,9 Ham1  

  SV2SNH Σελίδα Facebook:https://www.facebook.com/groups/215349699841903

               Γιά  πληροφορίες :  

       skype sakis: live:.cid.a4618de6d9be19be
       https://www.facebook.com/SV2SNH
       https://twitter.com/kroton000
       Telegram : @SV2SNH
  ----------------------------------------------------------            
  Σελιδα HELLAS-FRN:https://hellas-frn.weebly.com 
  Νεα σελιδα https://www.hellas-frn.net/ {υπο κατασκευη}
  HELLAS-FRN Admin:SV0XAL
  ***@***.***
            ------------------------------------------------
  Πρόσβαση στό δίκτυο  καί μέσω echolink:SV2SNH-R Ham1  Node:503858  KN10LN
  Προσβαση στο δικτυο  και μεσω echolink:SV2SNH-L Ham2  Node:499014  KM19TX
  Προσβαση στο δικτυο  και μεσω echolink:SV0XAL-L Ham3  Node:876693  KM18VH
  Προσβαση στο δικτυο  και μεσω echolink:SV8SYI-L Ham4  Node:925304  KM39GD
  Προσβαση στο δικτυο  και μεσω echolink:  ?   -L Ham5  Node:  ?
  Πρόσβαση στό δίκτυο  καί μέσω ZELLO,channel : HELLAS-FRN  --> Ham1
          
         ---------------------------------------------------
     Video σύνδεσης στο δικτυο HELLAS-FRN:
   https://www.youtube.com/watch?v=vGHUQaUi8js
   https://www.youtube.com/watch?v=0vjW-X9o8Wg

    Video χρήση HELLAS-FRN στο ΑΥΤΟΚΙΝΤΟ με την εφαρμογη GRN:
   https://www.facebook.com/groups/215349699841903/permalink/988641625846036/
        ----------------------------------------------------
  
        ----------------------------------------------------
[... 94 lines cut ...]
 13. OROPOS                 434.02500Mhz   T 71.9  SV0XAL Ham2
 14. THESSALONIKI           434.02500Mhz   T 71.9  SV2SNH Ham2
 15. VOLOS                  434.02500Mhz   T 71,9  SV4JYD Ham2
 16. AMOULIANI CHALK        145.28750Mhz   T 71,9  SV2SIG Ham1
 17. VOLOS PILIO             29.2500OMhz   T 71,9  SV4JYD Ham3
 18. KERKΥRA  [CORFU]       145.2875OMhz   T 71,9  SV8GTH Ham4
 19.?

  Αίτηση νέου κωδικού
  -------------------
  Σας ενημερώνουμε ότι μετά από προσπάθειες, διαφαίνεται πως το πρόβλημα της 
 μη αποστολής κωδικού οφείλεται στον Ευρωπαϊκό system manager.
 Δοκιμάζοντας διάφορα emails που ΔΕΝ είναι gmail, πήραμε σε όλα κωδικούς, 
 έναντι μηδενικής αποστολής στην περίπτωση των gmails.[NO gmails]
 Δεν γνωρίζουμε τι ιδιαίτερο πρόβλημα έχει με την Google o Ευρωπαϊκός,
 αλλά εφόσον αποστέλλει κωδικούς με αυτό τον τρόπο ακολουθούμε και οι υπόλοιποι.
 Μπορείτε λοιπόν να δημιουργήσετε κάποιο άλλο email και να ζητήσετε πάλι κωδικό 
 με την κανονική διαδικασία. Συνήθως εμφανίζεται στα ανεπιθύμητα.
 Δοκιμάστε αν μπαίνετε στο δωμάτιο test του Hellas-FRN και μετά ζητήστε την 
 πρόσβαση στα δωμάτια για ραδιοερασιτέχνες.
 Εννοείται ότι όσοι έχουν ήδη αποστείλει την άδειά τους και απλώς δεν μπορούσαν 
 να πάρουν κωδικό, μετά από αυτή τη διαδικασία απαιτείται μια απλή αποστολή του email.

73!
Hellas-FRN Administrator

dslotter, 2023-06-20:
The only FRN clients I could locate were for Windows and Android. This isn't applicable to Linux at this time, so closing.

closed 2023-06-20 by dslotter

sv2snh, 2023-06-21:
https://www.youtube.com/watch?v=AH14nHD3mhU&t=63s
https://frn4pi.org/content/changelog-download
https://www.youtube.com/watch?v=AH14nHD3mhU&ab_channel=Normsweb.com