dslotter/HamPi #231: 1920 X 1080 resolution output on Raspberry Pi

goaliebob99, 2024-08-01:
I would like to have the ham pi output in a 1080 X 1920 resolution from my PI 4.

jerryd, 2024-10-10:
I would like to second this. Go to wikipedia an look uo the standarda monitor resolutions. The ratio 800 x 600 is way off standards. It is a 5:3 ration.  Most modern monitors are a 16:9 or a 4:3.  The 1920 x 1080 is a 16:9 which would be very good.

JJTechman68, 2025-02-26:
I'd like to add my support for my sizes as well. I would like to fill the display on a large monitor.

tulipman, 2025-04-22:
Hi there.

Using the Hamclock since a week or so, great project, and it must have been a HUGE job to get it all the way to what it has become (v4.16). Compliments !!

I bought a 1920 x 1080 color LCD screen, but found out that only 3 resolutions are possible, and 1920 x 1080 is not one of them. I see quite a lot of messages in the ham-community that indicate that hams would like to have 1920 x 1080 resolutionn as an option.

73 de Robbert / TA2ZIX (aka PA3BKL) de Robbert